roblatour/ESP32TimeServer #4: Compile Errors

ianburton20, 2024-01-23:
Rob - thanks for this project. I get a couple of compile errors - any suggestions where I'm going wrong? Thanks

ESP32TimeServer_Test1.ino: In function 'void setDateAndTimeFromGPS(void*)':
ESP32TimeServer_Test1:355:19: error: aggregate 'tm wt' has incomplete type and cannot be defined
         struct tm wt;
                        ^~
ESP32TimeServer_Test1:369:34: error: 'mktime' was not declared in this scope
           candidateDateAndTime = mktime(&wt) + 1; // not sure why the + 1 but it is
                                                     ^~~~~

roblatour, 2024-01-23:
Hmmm, I just compiled it myself and it works fine.

Did you copy the code exactly? 

If yes, I'm guessing your issue is associated with a time library someplace on your system. 

I did a quick google and found a variety of people having the same problem with other projects.  Here are two examples: 
https://community.platformio.org/t/issue-with-struct-tm/19017/4
https://community.platformio.org/t/aggregate-tm-timeinfo-has-incomplete-type-and-cannot-be-defined/13715

Also, it if helps, the ESPTime.h library I am using is this one:
https://github.com/fbiego/ESP32Time
Originally, when I compiled my code I was using version 2.0.0 and it compiled fine.  Following your post I upgraded it to version 2.0.4 (the current version) and it also compiled fine.   Are you perhaps using an older library?

ianburton20, 2024-01-23:
Rob – Thanks. You have given me something definite to follow up. I’ll let you know how I get on. Thanks again – Ian

 

https://community.platformio.org/t/issue-with-struct-tm/19017/4
https://community.platformio.org/t/aggregate-tm-timeinfo-has-incomplete-type-and-cannot-be-defined/13715

https://github.com/fbiego/ESP32Time

ianburton20, 2024-01-24:
Rob

 

I’m OK with ESP32Time.h – using v2.0.4

 

But – I’m back into…

void setDateAndTimeFromGPS(void *parameter)

 

and that structure

struct tm wt;

 

error: aggregate 'tm wt' has incomplete type and cannot be defined

also….

 

I can’t find where mktime is declared in….

 

candidateDateAndTime = mktime(&wt) + 1; // not sure why the + 1 but it is

 

What version of  https://github.com/khoih-prog/Timezone_Generic

Are you using?

 

You #include <Timezone.h> but the library needs you to #include <Timezone_Generic.h>   - so I can’t fathom the version.

 

Thanks again.

Ian

 

https://community.platformio.org/t/issue-with-struct-tm/19017/4
https://community.platformio.org/t/aggregate-tm-timeinfo-has-incomplete-type-and-cannot-be-defined/13715

https://github.com/fbiego/ESP32Time

roblatour, 2024-01-24:
I just checked my libraries, it looks as if I'm now using
https://github.com/JChristensen/Timezone  (version 1.24)
for Timezone.h

Not sure when I switched to it, or why, but it seems to work with this project too.

roblatour, 2024-08-10:
closed due to lack of feedback

roblatour, 2024-08-10:
closed due to lack of feedback (no activity in over six months)

closed 2024-08-10 by roblatour