roblatour/ESP32TimeServer #5: TimeZone_Generic fails to install

Braehead, 2024-08-09:
I'm new to you ESP32TimerServer project and trying to get all the libraries loaded.  Only Issue I have is with TimeZone_Generic.  The library install fails over the wire library not available as a dependency (core of ESP32 and present on my system).  The TZ_G library is now Archived in GH and so issues can't be raised.  Wire is really now known as TwoWire (on most Arduino based boards) and this maybe where the issue lays.  I'm using Arduino 2.3.2 and ESP32 3.0.3.

Any suggestions of how to move forward?

roblatour, 2024-08-09:
The program is a year and a half old and it looks like it has run into a breaking change in the Arduino environment since then.

I ran some tests and changing the version of the esp32 board manager to 2.0.13 corrects the problem.

This is done in Arduino in the menu bar as follows:
Tools - Board: (whatever) - Boards Manager ...  - search for "esp32" (without the quotes) - esp32 by Espressif Systems - select 2.0.13 from the dropdown box - Update

when done exit and restart the Arduino program.

Please let me know if this fixes it for you, and if so I will update the comments in the program to mention this.

Braehead, 2024-08-10:
First thank you for coming back promptly.  Very much appreciated.  Sorry for the delay, I wanted to try some additional systems before detailing further findings.  And I know the issues of platform upgrades, sadly.

On the original machine.  Installing the Arduino ESP32 support (2.0.13) did not help.  Nor did this help on another system.  And I upgraded the Espressif platforms from 3.0.3 to 3.0.4.  Both the Arduino ESP32 and Espressif ESP32 systems have the Wire library as part of the core installs.  And I can include them into projects.  But trying to install the TZ_G library fails - "Failed to install. No valid dependencies solution found: dependency 'Wire' is not available.  I down graded my Espressif library to 2.0.13 as you suggested.  Restarted the IDE and still fails to install.  Versions verified.  Went back to 2.0.10 of Espressif ESP32 - still the same.  The error is immediate.  I have gone back to 3.0.4 as I have code to compile.  It's not the Arduino ESP32, as the first system does not have those boards loaded.

roblatour, 2024-08-10:
What a mess.  

However, I am using the Arduino IDE version 2.3.2 and as mentioned earlier the esp32 board manager to 2.0.13.

Maybe the version of the IDE is the difference?

Braehead, 2024-08-10:
OK.  Lets just get the versions and settings correct:

Arduino IDE 2.3.2 - check.

ESP32 Board library.  There are TWO ESP32 libraries that could be version 2.0.13.  The 'esp32' by Espressif and 'Arduino ESP32 Boards' by Arduino.  I presume we are talking about the Espressif boards definitions because that contains the 'Olimex ESP32 POE ISO'.  And the Arduino version does not.  I have both loaded due to also using Arduino based ESP projects.

I also presume the basic simple Olimex board as the compilation target?  As some have PSRAM and other specification variants.  Did some compile tests (without the TZ-G library available), and getting lots of unexpected errors.  I can't believe this much has changed since your original authorship took place.  So this has thrown further confusion into the mix.  But I'm ignoring these issues at the moment.  And focusing on why this TZ_G library refuses to install.  Wire is available in pretty much all MCU core board installs.  So why does it give an error.  I use I2C all over the place, and on various MCUs.  So it it present compiles and executes.

I think the author of TZ_G has archived all his GH publications because it blocks anyone raising an issue or discussing.  But it renders the project as 'dead', but downloadable.

I have a fairly vanilla Arduino IDE install on a Linux box which I will test in the morning.  And I will also look at VSStudio and PlatformIO too.

roblatour, 2024-08-10:
Yes, I am using the ESP32 by Espressif Systems board library.

Also, under the Arduino menu - File - Preferences - Additional Board Managers URLs I have the following links:
https://dl.espressif.com/dl/package_esp32_index.json
https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json

Here are the relevant settings under the Arduino menu - Tools:

Board: "OLIMEX ESP32-PoE-ISO"

Core Debug Level: "None"
Erase All Flash Before Sketch Upload: "Disabled"
Flash Frequency: 80MHz"
Partition Scheme: "Default"
Upload Speed: "115200"

Here is a screen shot of all the installed libraries that appear in Library manager (most have nothing to do with this project, but included here as it may be helpful)

[image]

Braehead, 2024-08-12:
OK, thanks for all this info, very much appreciated.  I have a clean install of the Ardunio IDE on a clean system.  So I'm building up and knocking down to get further along.  Clearly works for you, but not me.  Give me a few days to pin this down (or close to) and hopefully I'll get back to you with notes or whatever.

It is probably changes between IDE 1 & 2.  IDE 1 is looking OK with some avoidances, but IDE 2 keeps rejecting unless you came from IDE 1 with the library already installed.

roblatour, 2024-08-12:
Ok - please let me know if you would like me to send you / post the used libraries as listed here:
[image]

roblatour, 2024-08-13:
Glad to hear it.

I've found the Arduino platform increasingly frustrating over the years, for reasons such as this one.

Hope the program will be of good use to you.

Cheers!

Braehead, 2024-08-15:
OK.  Progress to some degree.

The issue is to do with the cited Timezone library: DO NOT USE the linked version Timezone_Generic by khoih-prog.  AKA Khoih Hoang.  Several issues:
- Khoih archives his publications to prevent issues being raised = No support
- This version of the library appears to be a rip-off of the original by  Jack Christensen.  Use Jack's version.
- The author cross referencies many other libraries that are published by himself or may have his input.  This results in any library from this author loading all the other libraries connected with Khoih.  For Timezone_Generic, this is an extra 20+ libraries that are not required.
- It is this huge dependancy list that screws up Adruino IDE1.x and 2.x.  Causing all kinds of errors when the dependency list is processed.

I have the ESP32Time server source compiling and linking under IDE 1.x.  IDE 2.x is giving me issues still.  However, I do not beleive the new issues are to do with Timezone, now I have ignored the  Khoih version and correctly using the  Jack Christensen version.

Many thanks @BigThunderSR for listing the file from PlatformIO that lists a far simpler dependency list that works on a vanilla Arduino install.  The IDE 1.x libraries work every time after a wipe down install of the IDE.  But as yet I have failed to clean install IDE 2.x or upgrade from 1.x to 2.x and it compile without an error.  I'm working on this as the next phase.  Now the Timezone library issue is resolved.

So not an Arduino issue.  Can anyone else do some tests?  IDE 1.x now gives me a WiFi UDP not defined issue and IDE 2.x give a Timezone function issue that I have not looked into at all at the point).  These faults are probably nothing to do with the Timezone library change over.  So remove all reference to Khoih libraries please.  But hold back on any documentation issues until I can clear errors with both IDEs please.

roblatour, 2024-09-06:
As this hasn't been updated in a while, what I've done is updated the code in the Arduino library to specify various Arduino Settings / Libraries to use.  I still haven't had any issues building myself, but hopefully this will help others.  As needed this issue can be reopened.

closed 2024-09-06 by roblatour